nfc-tools/libfreefare #37: Library aborts when malloc() fails

fuzxxl, 2015-05-21:
In commit f6c7f7692bdf9b5dc340d8efc72403b8027bfba7, code was introduced that calls `abort()` when `malloc()` fails in `mifare_desfire_crypto.c`. This is absolutely unacceptable. A library should under no circumstances terminate the program for such a minor problem as being out of memory. Please redesign the code so an error is reported by the library instead when `malloc()` fails to allocate memory.

smortex, 2015-05-26:
If we consider that [malloc never fails](http://stackoverflow.com/a/1440020) and that these `abort()` are only safekeepers, I would be okay to integrate a pull-request that handle this without aborting, but doing this would be at the lowest priority…

Please open a pull-request if you have a patch that fixes this issue for you.

closed 2015-05-26 by smortex

fuzxxl, 2015-05-26:
First of all, you handle failure to `malloc()` in all other places in the code, just not here. Why do you deviate from your perfectly fine error handling just in this one place? Second, the answer you linked is wrong and dangerous. It's pretty easy to get `malloc()` on Linux to fail, contrary to what this answer says. Also, Linux is not the only operating system in the world and `malloc()` may fail much more likely on other operating systems.

If you are unable to fix this issue on your own, I can write a patch for you, but you should really fix your own issues.

smortex, 2015-05-26:
> you handle failure to malloc() in all other places in the code, just not here

The code has been written in multiple shots over a long period of time, and while I agree with you that failing gracfuly is ways better than aborting, and consistency accross the code base would be better, the value of such a change is currently low from my point of view (I am not affected by this problem and there are other topics in the nfc-tools which are more important to fix before this kind of problems).

> If you are unable to fix this issue on your own, I can write a patch for you, but you should really fix your own issues.

As I said, this is at the lowest priority.  I would be happy to merge a pull request that fixes this if you can offer one, or will work on this if somebody can pay for this, but I can't offer another solution ATM due to time.

So, our possibilities are:

1. If we receive a pull-request referencing this issue, it will be merged;
2. If more people :+1: this issue, we will reopen it and reconsider this;
3. If none of the above happen, it is unlikelly to change, sorry.

Thanks!