nfc-tools/libfreefare #38: Timeout for nfc_initiator_select_passive_target

DasNaCl, 2015-07-24:
nfc_initiator_select_passive_target will block a whole thread, at least for me on a RaspberryPi B+ with standard but up-to-date configuration. Why is that so? Why does it wait infinitely until I hold a card in front of the sensor?

smortex, 2015-07-27:
Please use the mailing list or StackOverflow for questions/help, where a wider community will be able to help you. We reserve the issues tracker for issues only.

closed 2015-07-27 by smortex

DasNaCl, 2015-07-27:
I posted this here and not on a 'dedicated' platform for Q/A because of that: http://forums.nfc-tools.org/topic/1211/no-timeout-nfcinitiatorselectpassivetarget/
But okay, sorry for offtopic.

smortex, 2015-07-28:
This seems to be a libnfc problem, can you please open an issue on the proper project providing a sample application that reproduces the problem (or even better, a pull-request)?

Thanks!