nfc-tools/libfreefare #46: Please revert 1ce3db3c or make a new major release

fuzxxl, 2016-01-18:
The commit 1ce3db3c completely breaks the libfreefare API. This is a bad thing. Please either revert it or make a new major release to underline that the new version has a completely incompatible API.

smortex, 2016-01-18:
HI,

While it is not stated anywhere, me follow the [Semantic Versioning Specification](http://semver.org/).  Quoting it:
> 4 - Major version zero (0.y.z) is for initial development. Anything may change at any time. The public API should not be considered stable.

The referenced commit renames enum values.  If you do not want to synchronize your code (yet), I invite you to add some `#define`s to map the old names onto the new ones during the transition.

fuzxxl, 2016-01-18:
It has been like this for three years now without any indication that the software is still actively changing, at least not in any project documentation I found. No, not releasing a 1.0.0 version is not a magical ritual that deflects the responsibility for API stability from you. You are doing all your users a huge disservice by breaking the API like this. I'm seriously considering a fork at this point just so I don't have to fix my go bindings (which have been stable for two years now) every time you decide to break the API.

smortex, 2016-01-18:
I understand your point, I'm adding this issue to the 1.0.0 milestone

smortex, 2016-01-18:
As a consumer of the library, you may also have interesting insights about it.  Feel free to open issues or submit pull requests and assign them to the 1.0.0 release ;-)

fuzxxl, 2016-05-22:
If you are going to break the API anyway, consider removing all uses of pragma pack from the public interface. Pragma pack makes it very hard to use your library from other languages as not all foreign function interfaces support the non-standard pragma pack. There isn't a good reason why it's there either.

clausecker, 2024-04-20:
It seems like 42b21ff42f052b6f26aaf5e511e11f45b2a81b77 already broke the compatibility.