nfc-tools/libfreefare #54: Can we use these API using java on Raspberry pi? if yes then how?

annu0412, 2017-03-03:


SloCompTech, 2017-03-03:
This library is written in C/C++, but  I think it is posible to use it with java if you make module which communicate via D-Bus with your java application

annu0412, 2017-03-03:
Thanks Martin for reply.

I have PN532 i am using it over i2c, can you pls elaborate how can i use
libfreefare in java for this?

Thanks for your help.

Regards
Anurag

neomilium, 2017-03-03:
To use any library written in C/C++ (and some other languages) in Java :

https://en.wikipedia.org/wiki/Java_Native_Interface

You have to wrap native library using JNI and you will be able to use this
library as a Java class.

smortex, 2017-04-03:
Please use [Stack Overflow](http://stackoverflow.com/questions/tagged/nfc/) for questions/help, where a wider community will be able to help you. We reserve the *issues* tracker for **issues** only.

closed 2017-04-03 by smortex