nfc-tools/libfreefare #55: macOS and endian functions

bettse, 2017-03-04:
I was having some trouble with compiling the felica-read-ndef example when I ran `make`, and I also was getting some missing symbol errors [1] when executing `mifare-desfire-info` after adding a `mifare_desfire_read_data` call to the example.  After a bit of debugging, I think I've found the answer.  

## felica-read-ndef

### Problem

Uses `be32toh`, but only has conditional includes for `sys/endian.h` and `endian.h`.  I can't speak to if the examples _should_ be cross platform, but I figure if they're in the repo, probably?

### Solution

```
diff --git a/examples/felica-read-ndef.c b/examples/felica-read-ndef.c
index 55483d7..60a9593 100644
--- a/examples/felica-read-ndef.c
+++ b/examples/felica-read-ndef.c
@@ -30,9 +30,14 @@
 #  include <endian.h>
 #endif

+#if defined(HAVE_COREFOUNDATION_COREFOUNDATION_H)
+#  include <CoreFoundation/CoreFoundation.h>
+#endif
+
 #include <nfc/nfc.h>

 #include <freefare.h>
+#include "../libfreefare/freefare_internal.h"

 #define NDEF_BUFFER_SIZE 512
```

## Symbol not found: _le16toh

### Problem

This is a lot like #31 .  

`freefare_internal.h` uses `if !defined(le32toh) && defined(CFSwapInt32LittleToHost)` to determine if it needs to define the `le32toh` macro, but I was checking "/System/Library/Frameworks/CoreFoundation.framework/Versions/A/Headers/CFByteOrder.h" and (on macOS Sierra) it has `CF_INLINE uint32_t CFSwapInt32BigToHost(uint32_t arg) {`, which appears to be inlining as a function, instead of a macro, and I think that means `defined` doesn't work on it[2].  I say "appears", "think", etc because I do not know C or libfreefare very well, so there may be gaping holes in my logic or understanding.

### Solution

I just reused the same macro that was used to determine if the CoreFoundation header would be included in the first place.

```
diff --git a/libfreefare/freefare_internal.h b/libfreefare/freefare_internal.h
index b792ced..f7f8568 100644
--- a/libfreefare/freefare_internal.h
+++ b/libfreefare/freefare_internal.h
@@ -49,14 +49,14 @@
 #  define be16toh(x) betoh16(x)
 #endif

-#if !defined(le32toh) && defined(CFSwapInt32LittleToHost)
+#if !defined(le32toh) && defined(HAVE_COREFOUNDATION_COREFOUNDATION_H)
 #  define be32toh(x) CFSwapInt32BigToHost(x)
 #  define htobe32(x) CFSwapInt32HostToBig(x)
 #  define le32toh(x) CFSwapInt32LittleToHost(x)
 #  define htole32(x) CFSwapInt32HostToLittle(x)
 #endif

-#if !defined(le16toh) && defined(CFSwapInt16LittleToHost)
+#if !defined(le16toh) && defined(HAVE_COREFOUNDATION_COREFOUNDATION_H)
 #  define be16toh(x) CFSwapInt16BigToHost(x)
 #  define htobe16(x) CFSwapInt16HostToBig(x)
 #  define le16toh(x) CFSwapInt16LittleToHost(x)
```

[1]
```
dyld: lazy symbol binding failed: Symbol not found: _le16toh
  Referenced from: /usr/local/opt/libfreefare/lib/libfreefare.0.dylib
  Expected in: flat namespace

dyld: Symbol not found: _le16toh
  Referenced from: /usr/local/opt/libfreefare/lib/libfreefare.0.dylib
  Expected in: flat namespace
```

[2]https://gcc.gnu.org/onlinedocs/cpp/Defined.html
>test whether a certain name is defined as a macro

smortex, 2017-04-03:
Can you please fill-in a pull request?

bettse, 2017-04-03:
Sure: https://github.com/nfc-tools/libfreefare/pull/56

closed 2017-04-04 by smortex