nfc-tools/libfreefare #57: CMake find_library

swalkner-nxs, 2017-04-11:
Hello,

I was wondering if anybody builds with CMake? I'm having troubles when calling cmake with "-DDLIBNFC_INSTALL_DIR=<...path>".
When I change the find_library call in cmake/modules/FindLIBNFC.cmake to "nfc" (instead of "libnfc") - everything works out fine...

Thanks in advance,
stefan
```
diff --git a/cmake/modules/FindLIBNFC.cmake b/cmake/modules/FindLIBNFC.cmake
index f2b1b75..0c5d6ac 100644
--- a/cmake/modules/FindLIBNFC.cmake
+++ b/cmake/modules/FindLIBNFC.cmake
@@ -9,7 +9,7 @@ message("libnfc install dir: " ${LIBNFC_INSTALL_DIR})
 find_path(LIBNFC_INCLUDE_DIRS NAMES nfc/nfc.h PATHS ${LIBNFC_INSTALL_DIR}/include)
 message("libnfc include dir found:  " ${LIBNFC_INCLUDE_DIRS})
 
-find_library(LIBNFC_LIBRARIES libnfc PATHS ${LIBNFC_INSTALL_DIR}/lib)
+find_library(LIBNFC_LIBRARIES nfc PATHS ${LIBNFC_INSTALL_DIR}/lib)

```

smortex, 2017-04-11:
Yup, it seems to be somewhat broken.  May I ask you to fill-in a Pull-Request for this change?

swalkner-nxs, 2017-04-11:
Oh yes - I'll create one. Another question about the same topic:
why is there an explicit 'add_definitions("-std=c99")'
Is someone requiring that? This leads to compile errors (optind undeclared) on my Ubuntu 14.04
I commented that out - and everything works as expected...

smortex, 2017-04-11:
Well, the code use *moden* (sic) c99, e.g. variable declarations are not always at the beginning of a bloc ([example here](https://github.com/nfc-tools/libfreefare/blob/master/libfreefare/mifare_desfire_crypto.c#L116)); structure members are initialized the new way ([example here](https://github.com/nfc-tools/libfreefare/blob/master/libfreefare/mifare_desfire.c#L308)).  While most c89 compilers support c99 extensions, I think it's better to use `-std=c99` because it's the version of the language that was used and compilers defaults to c89.

Are you sure there is no previous error (you can copy & paste a full build error)? The name `optind` is linked to `getopt(3)` in my mind, and maybe some header is missing on your system…

swalkner-nxs, 2017-04-12:
Hello Again,

toyed around a bit now (concerning c99, etc). How about:
diff --git a/CMakeLists.txt b/CMakeLists.txt
index a3cb9bf..a3d76fe 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -3,7 +3,11 @@ cmake_minimum_required(VERSION 2.6)
 
 set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${CMAKE_CURRENT_SOURCE_DIR}/cmake/modules/")
 
-add_definitions("-std=c99")
+IF(CMAKE_C_COMPILER_ID STREQUAL "GNU")
+  add_definitions("--std=gnu99")
+ELSE()
+  add_definitions("-std=c99")
+ENDIF()

smortex, 2017-04-15:
Hum, took some time to have a look.  I booted-up an Ubuntu Xenial virtual machine and attempted to build the code.

CMake is definitively broken: the source files expect `config.h` to be present (it is generated by the `./configure` script generated by the autotools).  AFAICR, CMake was introduced at some point to permit build on Windows, which is not officially supported…  Some linting seems required (at least check if `HAVE_CONFIG_H` is defined before including `config.h`).

Regarding `getopts`, according the [the man page](https://linux.die.net/man/3/getopt), `getopts.h` is only required when using `getopt_long()` and `getopt_long_only()` which are not used.  This second problem therefore seems bogus, but maybe the `config.h` that was found and used in your case defines some macros that change what the preprocessor generates?  In this case, this second problem is a side effect of the first one.

I think the best is:

1. Have a clean repo (clone it or clean is using `git clean -fdx`);
2. Create an empty directory for the build and change to this directory;
3. Run `cmake /path/to/libfreefare-source` to generate the build files;
4. Build using `make`

This way you should hit failures that you can fix in the libfreefare code.  When everything is fine, please update your #58 Pull-Request :smile:

swalkner, 2017-04-19:
Hey,

Yes - you're right with your suggestions - but which GCC are you compiling with? (I'm using GCC-6.2.0)
1) config.h - generated one with 'configure' and tried to compile the source then - same issue with optind. Previously I 'touch'ed an empty config.h (thus I forgot all about it...)

2) Do you want me to add HAVE_CONFIG_H ifdef everytime where config.h is included? An option would be to have cmake generate an empty config.h (for the folks that are doing cmake building in Linux)

3) The optind issue stays with C99 and leads to undefined references with htobe16, le16toh, et al.

Since CMake is primarily used for windows building - maybe the GNU Check and gnu99 switch wouldn't be too bad?

Thanks

smortex, 2017-04-19:
This was tested with gcc 5.4.0, but the compiler should not pose any kind of problem with this code.

Regarding points 1 and 2, as CMake does not produce a `config.h` file, either such a file should be generated by CMake, or the `#include` should be "protected".  libnfc does both: here are links to where [CMake generates a `config.h` file](https://github.com/nfc-tools/libnfc/blob/master/CMakeLists.txt#L27), [defines `HAVE_CONFIG_H`](https://github.com/nfc-tools/libnfc/blob/master/CMakeLists.txt#L30) and an example of [include protection](https://github.com/nfc-tools/libnfc/blob/master/libnfc/nfc.c#L73-L75) (this should be in all files including `config.h`).

In fact, the autotools are in charge of checking that a number of headers are available in order for them to be included ([see this example](https://github.com/nfc-tools/libfreefare/blob/master/libfreefare/mifare_desfire.c#L34-L52)).  Since they are undefined, none of these headers are included, producing warnings and errors (e.g. endianness functions).  I guess that CMake should perform some checks and define macros accordingly.

The `-std=c99` flag makes the compiler more strict.  In order to detect problems, compile with extended checks, for example with these CFLAGS: `-Wall -Wextra -Werror -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=address` (All the `-Wno-error=...` are required because of the ugly macros).  I just checked (and fixed a [problem](https://github.com/nfc-tools/libfreefare/commit/3975019b3c5b10b7d676d8b0af726041d5003b37)): it now compiles with the autotools.

swalkner, 2017-04-21:
Sorry for the delay.
How about that:
https://github.com/swalkner/libfreefare/commit/21c585abf5195059d6e019bd666f81f96dd385eb
Built with autotools and cmake (on ubuntu 14.04)
stefan.

smortex, 2017-04-24:
Excellent! :shipit:

smortex, 2017-04-24:
Just merged #58, thanks!

closed 2017-04-24 by smortex