nfc-tools/libfreefare #60: Do not fail crypto error without reporting

manu0401, 2017-04-26:
Add last_pcd_error value for some crypto errors. Also enable the
warning even if WITH_DEBUG is not set, so that we have a clue of
why it failed.

[0001-Do-not-fail-crypto-error-without-reporting.patch.txt](https://github.com/nfc-tools/libfreefare/files/960039/0001-Do-not-fail-crypto-error-without-reporting.patch.txt)

smortex, 2017-04-27:
Makes sense!

May I ask you to keep the message printing within the `WITH_DEBUG` condition (writing to stderr from a library is not so good, there are a few locations where it's done in the library unfortunately… feel free to correct this if you wish :wink:).

Also, can you please [fill-in a Pull-Request](https://help.github.com/articles/github-flow/) so that we can benefit from GitHub's tools and review system?

Thanks!

manu0401, 2017-04-27:
I think I managed to do it (github documentation misses a few steps).
https://github.com/nfc-tools/libfreefare/pull/61

smortex, 2017-04-27:
Perfect, thanks!

closed 2017-04-27 by smortex