nfc-tools/libfreefare #69: Add safety check if lenght == 0

der-stefan, 2017-08-22:
https://github.com/nfc-tools/libfreefare/blob/69bf99b33daf350d23aee8abffa96b4ed96151d1/libfreefare/mifare_desfire_crypto.c#L70

We should add right after that:
```
if(len == 0) {
   return;
}
```
Else there occurs a segfault when reading `data[0]` and `data[-1]`. This happens, if a card isn't read correctly.

Same issue for function `lsl` and `xor` in the same file.

SloCompTech, 2017-08-24:
I agree, it needs to be handled in one way, but it is not needed in xor, because there is no len - 1

smortex, 2017-08-31:
:+1: May one of your create a Pull-Request please?

smortex, 2017-08-31:
Maybe with unit-test?  Our unit tests are more integration tests than anything else, so maybe it's okay do to small tests too :wink:

closed 2019-10-29 by smortex