nfc-tools/libfreefare #84: error: ‘NMT_BARCODE’ undeclared - when compinling

RichusX, 2018-03-25:
Trying to compile libfreefare on a Raspberry Pi, but encountering this issue. Any ideas?
```bash

make[2]: Entering directory '/home/pi/libfreefare/libfreefare'
  CC       mifare_key_deriver.lo
mifare_key_deriver.c: In function ‘mifare_key_deriver_update_uid’:
mifare_key_deriver.c:141:10: error: ‘NMT_BARCODE’ undeclared (first use in this function)
     case NMT_BARCODE:
          ^~~~~~~~~~~
mifare_key_deriver.c:141:10: note: each undeclared identifier is reported only once for each function it appears in
Makefile:605: recipe for target 'mifare_key_deriver.lo' failed
make[2]: *** [mifare_key_deriver.lo] Error 1
make[2]: Leaving directory '/home/pi/libfreefare/libfreefare'
Makefile:468: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/pi/libfreefare'
```

SloCompTech, 2018-03-25:
Hi.
do you use standard linux dist for Raspberry Pi ? Did you install `libnfc` first ? Did you install all required libraries ?

I would recommend you check following:
- update, upgrade os with `sudo apt-get update` and `sudo apt-get upgrade`
- check if you have `libnfc` installed, it is available [here](https://github.com/nfc-tools/libnfc)
- install required libraries: `libusb-dev`,`autoconf`,`automake`,`make`,`libtool`,`libssl-dev`,`pkg-config`,`libi2c-dev`

Here are my install commands which worked for me:
```
cd <your directory with library repo>
autoreconf -vis
./configure --prefix=/usr
make
sudo make install
```
If this doesn't work, report as many details as you can.

Hope it helps,
Cheers

RichusX, 2018-03-25:
Hey, yes I've completed all of the steps you've mentioned. I also attempted to delete libfreefare folder & pull new, but that produced the same error as in the original post.

SloCompTech, 2018-03-25:
Based on output above you should check your libnfc installation, because NMT_BARCODE is defined in [libnfc](https://github.com/nfc-tools/libnfc/search?utf8=✓&q=NMT_BARCODE)

RichusX, 2018-03-26:
`libnfc` seems to work fine from what I've seen. It finds i2c reader & reads cards fine. 

I also attempted to fix this by wiping the SD card, installing Raspbian and making the libraries again, but have the same error.

One thing I noticed when compiling `libnfc` is that it throws these warning in the middle, but continues fine: 
```bash
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/nfc-emulate-forum-tag4 /home/pi/libnfc/debian/tmp/usr/bin/nfc-emulate-forum-tag4
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/nfc-jewel /home/pi/libnfc/debian/tmp/usr/bin/nfc-jewel
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/nfc-list /home/pi/libnfc/debian/tmp/usr/bin/nfc-list
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/nfc-mfclassic /home/pi/libnfc/debian/tmp/usr/bin/nfc-mfclassic
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/nfc-mfultralight /home/pi/libnfc/debian/tmp/usr/bin/nfc-mfultralight
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/nfc-read-forum-tag3 /home/pi/libnfc/debian/tmp/usr/bin/nfc-read-forum-tag3
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/nfc-relay-picc /home/pi/libnfc/debian/tmp/usr/bin/nfc-relay-picc
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/nfc-scan-device /home/pi/libnfc/debian/tmp/usr/bin/nfc-scan-device
```
...
```bash
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/nfc-anticol /home/pi/libnfc/debian/tmp/usr/bin/nfc-anticol
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/nfc-dep-initiator /home/pi/libnfc/debian/tmp/usr/bin/nfc-dep-initiator
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/nfc-dep-target /home/pi/libnfc/debian/tmp/usr/bin/nfc-dep-target
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/nfc-emulate-forum-tag2 /home/pi/libnfc/debian/tmp/usr/bin/nfc-emulate-forum-tag2
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/nfc-emulate-tag /home/pi/libnfc/debian/tmp/usr/bin/nfc-emulate-tag
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/nfc-emulate-uid /home/pi/libnfc/debian/tmp/usr/bin/nfc-emulate-uid
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/nfc-mfsetuid /home/pi/libnfc/debian/tmp/usr/bin/nfc-mfsetuid
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/nfc-poll /home/pi/libnfc/debian/tmp/usr/bin/nfc-poll
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/nfc-relay /home/pi/libnfc/debian/tmp/usr/bin/nfc-relay
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/pn53x-diagnose /home/pi/libnfc/debian/tmp/usr/bin/pn53x-diagnose
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/pn53x-sam /home/pi/libnfc/debian/tmp/usr/bin/pn53x-sam
libtool: warning: '../libnfc/libnfc.la' has not been installed in '/usr/lib/arm-linux-gnueabihf'
libtool: install: /usr/bin/install -c .libs/pn53x-tamashell /home/pi/libnfc/debian/tmp/usr/bin/pn53x-tamashell

```

SloCompTech, 2018-03-26:
Check if you installed libnfc library, check dirs `/usr/lib`,`/usr/local/lib`,`/usr/local/lib/arm-linux-gnueabihf`.

RichusX, 2018-03-27:
I ran the dpkg install, but you're right, I can't see them...

```sh
pi@hermes:~ $ ls -alh /usr/lib | grep nfc
pi@hermes:~ $ ls -alh /usr/local/lib | grep nfc
pi@hermes:~ $ ls -alh /usr/local/lib/arm-linux-gnueabihf | grep nfc
ls: cannot access '/usr/local/lib/arm-linux-gnueabihf': No such file or directory
```

SloCompTech, 2018-03-27:
Did you install libnfc from source ? In this case was it latest repo ?

RichusX, 2018-03-27:
I did not and I think that was my mistake... Trying now and will report back on how it goes.

RichusX, 2018-03-27:
Unfortunately I am getting the same exact error.

RichusX, 2018-03-27:
HURRAY! Got it working by installing dev version of libnfc instead of latest stable as I was doing before.

closed 2018-03-27 by RichusX

tenznhok, 2019-01-23:
Hi, where is the INSTALL file for libnfc? I cant fine it anywhere.

tenznhok, 2019-01-23:
I use https://github.com/nfc-tools/libnfc/releases libnfc-1.7.1 and still have the same problem about 
```
mifare_key_deriver.c:141:10: error: ‘NMT_BARCODE’ undeclared (first use in this function)
     case NMT_BARCODE:
          ^~~~~~~~~~~
```

jdonald, 2019-07-22:
@tenznhok that release is more than four years old. You likely have to build libnfc off the tip of its master, and its procedure is not unlike that of libfreefare.

    git clone https://github.com/nfc-tools/libnfc
    cd libnfc
    autoreconf
    ./configure --prefix=/usr/local
    make
    sudo make install

Kyle-Falconer, 2020-03-30:
This problem is still present. I ran into it on my mac (macOS Catalina 10.15)

I was trying to install this from homebrew, but I needed libfreefare-dev, which isn't on homebrew.

I think this issue should be re-opened and some precompile flags should be added to inject a dummy `NMT_BARCODE` if it doesn't exist, since it seems libnfc-dev is also not on homebrew, and the version that's up there doesn't want to link correctly enough for libfreefare to be able to find the library files for proper linking.

Like the OP, the normal nfc tools were working for me prior to running into this issue.