nfc-tools/libfreefare #85: Can´t open device with freefare but nfc

rylox, 2018-04-11:
Hello,

I succesfully compiled the examples from libfreefare and wanted to test them. 
But everytime i ran the program i received : 
error   libnfc.driver.pn532_spi   Unable to wait for SPI data. (RX)
back.
The strange with is that I before tried to run the quick_start_example from libnfc, and it worked without a problem.
I´m using a I2C connection by the way (and yes i configured it all properly, or atleast thats what i think because nfc-list works)

nfc-list output :

nfc-list uses libnfc 1.7.1
error   libnfc.driver.pn532_spi   Unable to wait for SPI data. (RX)
error   libnfc.driver.pn532_spi   Unable to wait for SPI data. (RX)
NFC device: pn532_i2c:/dev/i2c-1 opened
NFC device: pn532_i2c:/dev/i2c-1 opened

So i changed the code of the mifare-ultralight-info example from freefare to open and init the devices the same way like in the quick-start example.
It works, the device gets opened and can be used.
BUT
The whole rest of the code doesn´t work.
I tried to fix it for over 5 hours yet didn´t succeed.
But i noticed some things : 

the freefare_get_tags function doesn´t work properly, no information from the tag is stored.
the freefare_get_tag_type function returns null

I would appreciate any kind of advice on how to fix either the connection problem or the tag data one.

SloCompTech, 2018-04-11:
Hi,
1. Try to run nfc-poll and read tag
2. Try to run ntag-detect to show tag type (this is also reading tag)
3. What data is not stored on the tag ? Was it suppose to write data to tag ?
4. Which type of tag you are using ?

Cheers

rylox, 2018-04-12:
1. worked
2. worked output : tag with uid (too lazy to type it) is a NTAG21x
3. By data i meant the uid
4. It´s a mifare ultralight c ISO 14443-3A  NTAG213

Small note on the side, I also had the make error when trying to compile libfreefare with the case NTM_BARCODE. I simply removed the case and could compile then without any problems. 
Could this be causing any error ?

rylox, 2018-04-12:
Update

I no longer get the error after re-configuring the device.
I´ve set it manually now.
But when i use the mifare-ultralight-info example, i dont get any message back and the programm just exits without printing anything out.

rylox, 2018-04-12:
Another update

Ok the issue now is that the programm doesn´t recognise my tag as a ultralightc tag.
I made a small check with taste_mifare_ultralight_c, and it returned false.
The problem is that it´s for 100% sure a ultralight c tag, i checked it on the datasheet and with nfc-tools(a really usefull Android app to read / write tags).

Any idea why it wouldn´t recognise the tag as ultralight c ?

SloCompTech, 2018-04-12:
Try to scan tag with mobile phone with app nxp tag info and post screenshots.

rylox, 2018-04-12:
[image]
[image]
[image]
[image]
[image]

rylox, 2018-04-12:
Could you please run the current example mifare-ultralight-info ?
Just to confirm that the code is working.

SloCompTech, 2018-04-12:
I scanned one of my NTAG213 tags, properties match to my tag properties, to use this tag use ntag21x commands, but this tag is very strange, because this tag looks like to be super locked or corrupted, because with my tag also displays p as password protected.

rylox, 2018-05-04:
The ntag commands work without a problem, thank you for the help.

closed 2018-05-04 by rylox