nfc-tools/libfreefare #86: Write and read NDEF data on ntag21x

rylox, 2018-05-04:
Hello,
is there a way to convert raw byte data into NDEF data and write it on a tag using only the ntag21x commands or read NDEF data, again using only the ntag21x commands, and convert it back into raw byte data ?

SloCompTech, 2018-05-04:
Hello,
[here](https://github.com/nfc-tools/libfreefare/blob/master/libfreefare/freefare.h#L524) are basic commands for creating TLVs to create containers with your data, but for reading with this you would need to read whole data block or proces from start and read as you go, you can also look at:
- https://github.com/nfc-tools/libndef
- https://github.com/haldean/ndef
- https://learn.adafruit.com/adafruit-pn532-rfid-nfc/ndef
- https://stackoverflow.com/questions/37220910/how-to-interpret-ndef-content-on-mifare-classic-1k

Hope it helps

rylox, 2018-05-04:
Thanks for the lightning fast answer :D

rylox, 2018-05-07:
I´ve written a function to write the ndef data using the ntag21x command, but the strange thing with it is that it always puts the number 0xFF at the end of the message, so the its type/message/0xFE/0xFF.

Any idea why so ? 

Here´s the code : 

`#define ARRAY_SIZE(x) sizeof(x)/sizeof(x[0])

const uint8_t ndef_default_msg[] = {0x67, 0x6f, 0x6f, 0x2e, 0x67, 0x6c};
const uint8_t block[] = {0x06, 0x07, 0x08, 0x09};

int write_ndef(FreefareTag tag, uint8_t *data, const uint8_t type, size_t isize) {

    uint8_t *ndef_msg;
    size_t  ndef_msg_len;
    int sector_count;

    ndef_msg = data;
    ndef_msg_len = isize;

    uint8_t write_data [4];

    printf("Printing raw message :\n");
    print_hex(ndef_msg, ndef_msg_len);

    size_t encoded_size;
    uint8_t *tlv_data = tlv_encode(type, ndef_msg, ndef_msg_len, &encoded_size);

    printf("NDEF file is %zu bytes long.\n", encoded_size);

    printf("Printing NDEF formatted message :\n");
    print_hex(tlv_data, encoded_size);

    sector_count = encoded_size / 4;
    if((encoded_size%4)!= 0)
    sector_count++;

    for (size_t i = 0; i < sector_count; i++) {

        for (size_t f = 0; f < 4; f++) {

            if((i * 4 )+ f > encoded_size) {
                write_data[f] = 0x00;
            }

            else {
            write_data[f] = tlv_data[(i * 4) + f];
         }
     }

        ntag21x_write(tag, block[i], write_data);
    }

    return 1;
}`

SloCompTech, 2018-05-07:
TLV ends with 0xfe or 0xff.

rylox, 2018-05-07:
0xfe

SloCompTech, 2018-05-07:
Could you please also post program output.

Check Section TLV Blocks -> Length field https://learn.adafruit.com/adafruit-pn532-rfid-nfc/ndef could ff mean start of newTLV with 3byte format with len 0?

rylox, 2018-05-07:
[image]

rylox, 2018-05-09:
found it, it was a missing `=` in the if statement of the second for loop.
Works now all fine.

closed 2018-05-09 by rylox

SloCompTech, 2018-05-09:
Great to hear 👍 
Would you be prepared to create an example and contribute it to this repo ?

SloCompTech, 2018-05-09:
Would be nice if you could also specify type of data like URL, email ... 

Btw Thanks for your hard work