nfc-tools/libfreefare #87: How I can Desfire authenticate in two steps?

crossmax, 2018-05-16:
Hi.
I'm trying authenticate on mifare desfire EV1 with a key generating for a external device (SAM module).
For that, in my old code where I used native commands, I send 0x0A command to PICC, and I get a random number within response. With this number, the SAM generates the key, and I send 0xAF command and 16 bytes for the key to resume authentication with the PICC.
With libfreefare functions, I don't know how to authenticate in two steps.

The output debug using libfreefare is as follows but I need to have control of the two steps of authentication:

```
*** mifare_desfire_select_application ***
===> 0000   90 5a 00 00 03 00 00 01 00 
<=== 0000   91 00
PICC replied OPERATION_OK
*** authenticate ***
===> 0000   90 0a 00 00 01 02 00
<=== 0000   6c 62 96 21 59 07 b6 4e 91 af

### I need stop here and do something ###

*** authenticate ***
===> 0000   90 af 00 00 10 a3 54 d9 35 6f e3 3e 0d 68 a5 31
===> 0010   bb 81 40 4b f6 00
<=== 0000   91 ae
PICC replied AUTHENTICATION_ERROR
```
How I can do it? Thanks

SloCompTech, 2018-05-18:
Could you please provide source code for this section ?

crossmax, 2018-05-21:
Hi @SloCompTech 
I'm testing with this:
`   MifareDESFireAID aid = mifare_desfire_aid_new(0x00010000);
    res = mifare_desfire_select_application(tag, aid);
    cut_assert_success("mifare_desfire_select_application()");

    uint8_t s, c;
    res = mifare_desfire_get_key_settings(tag, &s, &c);
    cut_assert_success("mifare_desfire_get__key_settings");
	
	uint8_t v;
    mifare_desfire_get_key_version(tag, 2, &v);
    cut_assert_success("mifare_desfire_get_key_version");

    //mifare_desfire_auto_authenticate(tag, 2);
    MifareDESFireKey key1 = mifare_desfire_des_key_new_with_version(key_data_null);
    res = mifare_desfire_authenticate(tag, 2, key1);
    cut_assert_success("mifare_desfire_authenticate()");`
I think is not possible use libfreefare authenticate function  with a external generate key. I'm going to modify the function to divide it into two phases. The only question I have is when should I make calls to `mifare_cypher_blocks_chained` function, I'm not sure if I call it every time I call to `DESFIRE_TRANSCEIVE` function.
Any suggestion is very welcome!

SloCompTech, 2018-05-21:
As far as I understand authentication with Mifare Desfire  you need to authenticate with application key after you select app, additionaly you authenticate with PICC key before you create, delete app, so I assume authentication needs to be preformed once, but I'm not 100% positive.

smortex, 2018-05-22:
@crossmax I think this is currently not supported but would be an interesting contribution (i have never worked with a SAM, all devices I had had no SAM)!

The `authenticate()` function does the full handshake as a single step:
https://github.com/nfc-tools/libfreefare/blob/c18f7028406e3de2938cbab070539f6d6c031baf/libfreefare/mifare_desfire.c#L350-L434

I think that if you split this function into smaller pieces, a lot of code could be shared between authenticating with a MifrareDESFireKey and against a SAM module.  Before starting this, you may want to be sure that if the communication is fully maced /  encrypted, not *every* command has to pass through the SAM…  If it is the case, I guess that updating the ugly `DESFIRE_TRANSCEIVE*` macro will be required, and last time I tried to convert them to functions, I failed to do so (that leads to **a lot more** changes than what I expected).

crossmax, 2018-05-29:
@smortex I was doing what you said and two steps authentication works nice.
Wherever, I got involved in this modification because the Desfire reading times are too slow and i don't know how I to improve it.
For read a 144 bytes Desfire file my hardware takes about 100ms, that is 11kbps and the time is same if I use NBR_106 or NBR_847 baud rate.

Are this times normal? With other hardware and the same Desfire cards but using NXP native commands I was getting times almost twice as fast.
Can it be due to apdu encapsulation? Or be due to poor quality reader??
Any suggestion to improve reading times is welcome!!!

SloCompTech, 2018-05-29:
@crossmax which bus are you using ? I2C ? SPI ? UART ?

crossmax, 2018-05-29:
@SloCompTech I'm using SPI

SloCompTech, 2018-05-29:
In connection string you can set speed, `pn532_spi:/dev/spidev0.0:1000000` is working for me, speed is set after `:` in bps.

crossmax, 2018-05-29:
Yes. I checked the settings first of all. Up to 3,5MHz works fine but reading speed is still slow

smortex, 2018-05-29:
Hi!

For debugging timing problems, I usually end-up using [DTrace](https://en.wikipedia.org/wiki/DTrace) (I mainly run FreeBSD).  If you are running on GNU/Linux, I guess that [looking into this](http://www.brendangregg.com/FlameGraphs/cpuflamegraphs.html) (super-complete, by some NetFlix staff) or [this](http://nanxiao.me/en/use-perf-and-flamegraph-to-profile-program-on-linux/) (same tools, in a TL;DR fashion) can help?

This should help you to measure which step is slow… processing data, sending it to the device, reading, etc.

crossmax, 2018-05-30:
Hi,
I've never used DTrace, but I'm testing with libfreefare's debug mode and linux timestamp (timeval). Will I get more information if I use DTrace? 
These are the traces where you can see the time it takes to read the file:

    *** mifare_desfire_select_application ***
    ===> 0000   90 5a 00 00 03 00 00 01 00
    <=== 0000   91 00
    PICC replied OPERATION_OK
    Time elapsed after the SelectApp 0.054548s

    *** authenticate_step1 ***
    ===> 0000   90 0a 00 00 01 02 00
    <=== 0000   84 79 5f 8f cc b5 d9 f7 91 af

    *** negotiation with sam ***

    *** authenticate_step2 ***
    ===> 0000   90 af 00 00 10 10 be de d5 e6 7f 45 91 19 1e 33
    ===> 0010   da 5f 91 94 0a 00
    <=== 0000   74 22 f0 69 33 19 12 67 91 00
    PCD  0000   bd ce 76 e8 05 00 00 50
    PICC 0000   b5 1e f1 c8 0b 93 71 e1
    PICC replied OPERATION_OK
    Time elapsed after authentication 0.137789s

    *** mifare_desfire_get_file_settings ***
    ===> 0000   90 f5 00 00 01 00 00
    <=== 0000   01 03 40 24 8d 00 00 91 00
    *** read_data ***
    ===> 0000   90 bd 00 00 07 00 00 00 00 00 00 00 00
    <=== 0000   09 15 22 e7 55 59 43 5a 0a 89 67 8c fb 93 cb c8
    <=== 0010   fe a0 f5 7e d0 e8 6d 75 c2 c3 a9 0d c7 fe 4d 96
    <=== 0020   6f 2d 61 2c 7e 49 7b 4e 53 06 98 19 13 16 af cc
    <=== 0030   a5 af b7 2c 33 4f 74 7c 91 af
    *** read_data ***
    ===> 0000   90 af 00 00 00
    <=== 0000   bf 71 d7 64 d7 24 a7 ce a4 1a ff 94 69 21 68 58
    <=== 0010   51 5e 45 3e ff ab ee 3e f7 f5 c9 22 00 d3 6c 2e
    <=== 0020   cc b6 89 bc f3 4b 45 e3 c9 55 43 86 ec af 8c 96
    <=== 0030   61 fd fb 45 a4 67 8f a4 91 af
    *** read_data ***
    ===> 0000   90 af 00 00 00
    <=== 0000   aa 7f b1 50 a4 b8 1f 61 20 35 77 0d 39 7b 0c 14
    <=== 0010   9a 10 43 fc 92 d6 70 e1 11 57 97 fd 0a bc 6a 31
    <=== 0020   91 00
    PICC replied `OPERATION_OK
    Time elapsed after read file (144bytes) 0.102637s

According to these traces everything is fine.
Could someone tell me how long it takes to read about 144 bytes of a Desfire EV1?

    ===> Version information for tag 047XXXXXXXXXXX:
    UID:                      0x047XXXXXXXXXXX
    Batch number:             0xba4450e390
    Production date:          week 11, 2013
    Hardware Information:
        Vendor ID:            0x04
        Type:                 0x01
        Subtype:              0x01
        Version:              1.0
        Storage size:         0x18 (=4096 bytes)
        Protocol:             0x05
    Software Information:
        Vendor ID:            0x04
        Type:                 0x01
        Subtype:              0x01
        Version:              1.4
        Storage size:         0x18 (=4096 bytes)
        Protocol:             0x05

smortex, 2018-06-07:
DTrace will not be available on Linux, but with gperf you should be able to gather information about *where* time is spent (e.g. your application; libfreefare; libnfc; or the kernel itself).

Once you have measured which part is slow, you will know where to see if it's possible to improve this.