nfc-tools/libfreefare #88: build error : error: ‘NMT_BARCODE’ undeclared (first use in this function)

davidhuwyler, 2018-05-22:
Hi

the make command gives me the following error:
`mifare_key_deriver.c: In function ‘mifare_key_deriver_update_uid’:
mifare_key_deriver.c:141:10: error: ‘NMT_BARCODE’ undeclared (first use in this function)
     case NMT_BARCODE:
          ^
mifare_key_deriver.c:141:10: note: each undeclared identifier is reported only once for each function it appears in
Makefile:605: recipe for target 'mifare_key_deriver.lo' failed
`
(autoreconf -vis and  ./configure --prefix=/usr run without problems)

Any ideas? 
Or is this a known problem?

smortex, 2018-05-22:
Looks like this was introduced in libnfc by this commit: https://github.com/nfc-tools/libnfc/commit/dcdbff0705c8eabb8f055e8e7b9ef237430d6577

This was never released in the libnfc project.  I guess you are not running the latest libnfc code. Install libnfc from the *master* branch and you should be on track.

smortex, 2018-05-25:
Close cause it's a dup of #83 and #84.

closed 2018-05-25 by smortex

davidhuwyler, 2018-05-28:
Thanks D3cedut0! Your solution works