nfc-tools/libfreefare #89: mifare_ultralight_connect fails when doing MIFARE_ULTRALIGHT tag

bobpombrio, 2018-05-26:
Not a MIFARE_ULTRALIGHT_C but the other one, the red headed step child. 

When I run the example : ./mifare-ultralight-info I get a response: Tag with UID 04087a040415d2 is a Mifare UltraLight
 Which is great but nowhere in the example code is there any other work done with it other the the _C.  But I am not a _C card.
SO I tried to write my own code but I used everything from the info code to create mine.

Any help with reading my tag would be appreciated.

Here ya go in it's glory.

#include <err.h>
#include <netdb.h>
#include <netinet/in.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

#include <freefare.h>
#include <nfc/nfc.h>

// globals
int sockfd;
int portno = 16783;
char *host = "192.168.1.9";

// Prototypes
int Net_Start(void);

// Code
int main(int argc, char *argv[]) {
  nfc_device *device = NULL;
  FreefareTag *tags = NULL;
  int error = EXIT_SUCCESS;

  // Start the network and connect to the server.
  //    if (Net_Start()) {
  //        printf("Network error.\n");
  //        exit (1);
  //    }

  nfc_connstring devices[8];
  size_t device_count;

  nfc_context *context;

  nfc_init(&context);
  if (context == NULL)
    errx(EXIT_FAILURE, "Unable to init libnfc (malloc)");

  device_count =
      nfc_list_devices(context, devices, sizeof(devices) / sizeof(*devices));
  if (device_count <= 0)
    errx(EXIT_FAILURE, "No NFC device found");

  for (size_t d = 0; d < device_count; d++) {
    if (!(device = nfc_open(context, devices[d]))) {
      warnx("nfc_open() failed.");
      error = EXIT_FAILURE;
      continue;
    }

    if (!(tags = freefare_get_tags(device))) {
      nfc_close(device);
      errx(EXIT_FAILURE, "Error listing tags.");
    }

    FreefareTag ult = NULL; // = tags[i];

    for (int i = 0; (!error) && tags[i]; i++) {
      switch (freefare_get_tag_type(tags[i])) {
      case MIFARE_ULTRALIGHT:
        //            case MIFARE_ULTRALIGHT_C:
        ult = tags[i];
        printf("We are doing ULTRALIGHT!\n");
        break;
      default:
        continue;
      }

      if (ult == NULL) {
        nfc_close(device);
        errx(EXIT_FAILURE, "The tag is crap.");
      }

      char *tag_uid = freefare_get_tag_uid(tags[i]);
      printf("Tag with UID %s is a %s\n", tag_uid, freefare_get_tag_friendly_name(tags[i]));

      // FreefareTag ult = tags[i];

      if (!mifare_ultralight_connect(ult)) {
        nfc_close(device);
        errx(EXIT_FAILURE, "Can not connect to ultralight");
      }

      unsigned char rd[4];

      if (!mifare_ultralight_read(ult, 4, &rd)) {
        nfc_close(device);
        errx(EXIT_FAILURE, "Can not read from ultralight");
      }

      printf("%X\n", rd[0]);
      printf("%X\n", rd[1]);
      printf("%X\n", rd[2]);
      printf("%X\n", rd[3]);

      if (!mifare_ultralight_disconnect(ult)) {
        nfc_close(device);
        errx(EXIT_FAILURE, "Can not disconnect from ultralight");
      }
    }

    nfc_close(device);
  }

  // successful exit
  printf("All done.\n");
  return 0;
}

int Net_Start(void) {

  struct sockaddr_in serv_addr;
  struct hostent *server;

  /* Create a socket point */
  sockfd = socket(AF_INET, SOCK_STREAM, 0);

  if (sockfd < 0) {
    perror("ERROR opening socket");
    return (1);
  }

  server = gethostbyname(host);

  if (server == NULL) {
    fprintf(stderr, "ERROR, no such host %s\n", host);
    return (1);
  }

  bzero((char *)&serv_addr, sizeof(serv_addr));
  serv_addr.sin_family = AF_INET;
  bcopy((char *)server->h_addr, (char *)&serv_addr.sin_addr.s_addr,
        server->h_length);
  serv_addr.sin_port = htons(portno);

  /* Now connect to the server */
  if (connect(sockfd, (struct sockaddr *)&serv_addr, sizeof(serv_addr)) < 0) {
    perror("ERROR connecting");
    return (1);
  }

  printf("Connected to %s\n", host);

  return (0);
}

Thanks,
Bob

SloCompTech, 2018-09-08:
Hey,
sorry for late reply. Could you please describe problem more ?

- What program prints out ?
- If I understood correctly, you have problem, that you don't know how to read & write ti Mifare Ultralight tag?
- If you have NFC enabled phone, it would be nice to attach a screenshot of NfcTools & Mifare Info apps

Cheers

bobpombrio, 2018-09-08:
I've moved onto a new project and I do not remember the problem.  Thanks for the reply though.

SloCompTech, 2018-09-09:
@smortex please close this issue.

closed 2018-09-09 by smortex