nfc-tools/libfreefare #100: Timeout should be configurable

broth-itk, 2019-01-23:
Dear freefare Team!

This is my first contribution so please don't blame me :)
I tried to push a fix to the modified timeout patch.

The initial patch from manu0401 sets a fixed timeout of 500ms for desfire operations.
See https://github.com/manu0401/libfreefare/commit/3e7036ce412dff74f532f1ad5a5c4736ba5bb5c7 for details.

This might work fine for most operations.
Formatting the desfire PICC can take more time and causes the operation to timeout.

After that, the tag is not operational, any new desfire operation (like selection) will result in a timeout.

With attached patch I added a new function "freefare_set_tag_timeout" which allows to set a custom timeout.

I thought I ruined six desfire tags when I first discovered that problem.

After setting a timeout of 1500 ms (instead of the previous fixed 500ms), I was able to recover all tags successfully.

I kindly ask to consider that patch for future addition to libfreefare.

If my patch happens to be incorrect (bad name, bad formatting etc.) please let me fix it.

Question: What is required in order to be allowed to officially contribute to the project?
It would be very easy to push my local copy of freefare :)

Thank you!
Best regards,

Bernhard

[freefare_set_timeout.diff.txt](https://github.com/nfc-tools/libfreefare/files/2786946/freefare_set_timeout.diff.txt)

smortex, 2019-01-24:
For easy review & comments, please submit a Pull Request instead of attaching a patch to an issue.  This allows up to review the changes in context, add in-line comments, and so on.  We can guide you to contribute if needed be.

If you are new to GitHub and need some help to start contributing with Pull Requests, please refer to this help page:
https://help.github.com/articles/creating-a-pull-request/

The preferred method is:
1. For the repository, download your fork (`git clone url-of-your-fork`)
2. Create a _feature-branch_ (`git checkout -b my-new-feature`)
3. Add commits (`vim`, `git add ...`, `git rm ...`, etc)
4 Push your branch to your fork of the repository (`git push -u origin my-new-feature`)
5. Open a Pull Request

broth-itk, 2019-01-24:
@smortex Thank you very much!

Who is responsible to add permissions to the repository?
Currently permission is denied to my user broth-itk to push my branch to the repo.

SloCompTech, 2019-01-24:
@broth-itk here are some guidelines how to add your changes:

- fork this repository to your own account (so you will have permission to push to the repo, fork it via github, it has nice fork button in the upper right corner)
- push changes to repo in your github account
- then at the top you will have button create pull request
- then at the left side shoud be original repo at the right your repo and branch that you want to merge here
- add description ....
- then submit pull request

broth-itk, 2019-01-25:
Thanks for your help! A new pull request is online. I hope it's all right.

closed 2019-01-25 by broth-itk