nfc-tools/libfreefare #106: mifare_ultralightc_authenticate function isn't working as expected

bluetiger30, 2019-02-18:
Hi , Thanks for this great library.

so I changed the 1st byte of the key from 0x49 to 0x48 in the mifare-ultralight-info which should tell me that authentication has failed because its not the correct key, however, it returns success !!!

if i change 0x49 to x40 it returns fail, so I'm not sure what is happening with the authenticate function

any idea on why it doesn't always provide the correct result?

graynk, 2019-06-10:
It will always return success after auth, but the following commands will yield error if the key was wrong, so you won't be able to read anything. I believe this is the problem of the card itself, not the library. You could check error status two times to circumvent the issue.

darconeous, 2019-10-29:
The least-significant bits of the 128-bit representation of the 2K3DES key are unused by the MIFARE Ultralight C.

Remember that DES uses 56-bit keys, but convention is to give the key in 64-bits—using the least significant bits of each byte as a parity check bit. The MIFARE Ultralight C ignores the parity bits.

This is why changing 0x49 to 0x48 doesn't cause the authentication to fail, but 0x49 to 0x40 does: in the former case you aren't actually changing any bits of the key, you are just changing the (unused) parity bit from `1` to `0`. In the later case you are actually changing the bits of the key and the authentication handshake fails accordingly.

darconeous, 2019-10-29:
Bottom line: `mifare_ultralightc_authenticate` behaved correctly in both cases.

closed 2019-10-29 by darconeous