nfc-tools/libfreefare #108: Docs: application creation on a desfire card is not clear

ericgoe, 2019-04-11:
Hello,
I have tried so far some functionalities of this library. I am not certain about the creation of an application. I try to explain what this code is doing. If I am wrong please correct me.
`MadAid mad_aid = { 0xB8, 0x57 };`
`MifareDESFireAID aid = mifare_desfire_aid_new_with_mad_aid(mad_aid, 0x5);`
`mifare_desfire_create_application(DESFIRE_TAG, aid, 0xFF, 0x1);`

In the first line I specify the functionality of the application (0xB8) and the pointer to the application (0x57) which I'll create.

In the second line I create a AID with the previous information and 0x5.
The documentation states that: n = 0x5: "n as the last nibble of the new AID.".

The third line actually creates the application. The document says that 0xFF is used as settings key settings and 0x1 is the number of the authentication key which will be used.

What defines the last nibble of an AID ? What can I do with it ?
What can I choose as key settings ? What are the possiblities ?
When I am uploading a key. Is this key marked with an number and can I use one key for multiple applications when defining that ?

SloCompTech, 2019-04-11:
Hello,
I don't have all the answers, but this is what I learned so far working with nfc:
- key settings: you can have multiple keys some with r & w premissions others with r permission
- you probably can use same key for different applications, but probably you need to set it for each application
- I recommend that you find documentation for specific tag type you are trying to use (there you can find available settings, bit mapping ...)
Cheers

ericgoe, 2019-04-12:
Hello SloCompTech,
thank you for you quick response.

I assume that the different key settings can be found too in the corresponding tag documentation.

To be honest, I didn't find yet the documentation for the Mifare DESFire EV1 tag. Where can it be found ?

Kind Regards

SloCompTech, 2019-04-12:
[mifaredocs.zip](https://github.com/nfc-tools/libfreefare/files/3073593/mifaredocs.zip)
Here are most of docs I found.

Yes, key settings are described in docs.

ericgoe, 2019-04-12:
Hello SloCompTech,

I will read me through these documentations.

Thank you for your help
Kind regards.

closed 2019-04-12 by ericgoe