nfc-tools/libfreefare #111: Supporting card with writable sector 0?

mandrean, 2019-05-08:
It seems like `libfreefare` assumes sector 0 is always read-only:
https://github.com/nfc-tools/libfreefare/blob/master/libfreefare/mifare_classic.c#L712

https://github.com/nfc-tools/libfreefare/blob/master/libfreefare/mifare_classic.c#L585

But there is also a special type of card with a writeable sector 0.

Does this mean `libfreefare` can't be used with those cards for writing to sector 0?

SloCompTech, 2019-05-09:
Hi,
you are probably correct about writing to sector 0 on mifare classic, probably this is safety feature so you don't write to sector 0 by accident, could you tell me more about this special card with writeable sector 0 ?

mandrean, 2019-05-09:
For example:

https://www.ebay.com/itm/Chinese-Magic-Mifare-Classic-1K-NFC-Tag-with-changeable-rewritable-UID/153477540176?hash=item23bbf96950:g:XIoAAOSw1AZchthD

or

https://www.ebay.com/itm/5PCS-UID-Changeable-Sector-0-Block-0-Writable-13-56Mhz-RFID-Proximity-Smart-Card/201930523372?hash=item2f03ff42ec:g:jHUAAOSwVEdbPe50

Also note that nfc-tools's `mfclassic` tool supports these cards:

http://manpages.ubuntu.com/manpages/bionic/man1/nfc-mfclassic.1.html

>The W option allows writing of special MIFARE cards that can be 'unlocked' to allow block 0 to be overwritten. This includes UID and manufacturer data. Take care when amending UIDs to set the correct BCC (UID checksum). Currently only 4 byte UIDs are supported.

>*** Note  that W and R options only work on special versions of MIFARE 1K cards (Chinese clones).

SloCompTech, 2019-05-09:
I reviewed library source code and I found that you could posiblly write to sector 0.

https://github.com/nfc-tools/libfreefare/blob/master/libfreefare/mifare_classic.c#L712:
- here if clause only protects sector 0 from formating (because it is usually only read-only)
https://github.com/nfc-tools/libfreefare/blob/master/libfreefare/mifare_classic.c#L585
- here if only causes function to immediatly return -1 that means that sections is locked (even if it isn't in your case)

But to be for sure, I suggest you try this with blank card and report result here, so we can solve this.

mandrean, 2019-05-10:
Yep, I found those two LoC when I was peeking around as well... They seem discouraging, but I will try it out!

smortex, 2019-05-18:
The idea of this library is to make it easier to manipulate cards by providing high-level API.  Therefore, it implements what cards are _supposed_ to do, and although these Chinese cards have some "extensions", they are not coping with the specifications of the original cards.

So libfreefare will ensure that your code with work reliably and consistently with all cards, with and without these "extensions".  If you need to rely on these extensions, I guess the simplest would be to fork the library and remove these special checks.

If you want this to be in libfreefare itself, I guess we can make this optional (add a configure flag to disable the extra checks), but please leave this disabled by default :smiley: 

I'll close this for now since usually, people who want to write sector 0 are not writing some "real world application" but are rather doing a proof of concept, and this is not the purpose of libfreefare.  If you really want to make this feature available as a configure option, feel free to open a Pull Request.

Thanks!

closed 2019-05-18 by smortex