nfc-tools/libfreefare #112: ntag21x_is_auth_supported command

shad0w82, 2019-06-07:
I'm having a hard time trying to understand the function ntag21x_is_auth_supported, from which the library can discern the tag type. 
The procedure is transceiving bytes 60 00, but nowhere on NTAG213/15/16 datasheet (https://www.nxp.com/docs/en/data-sheet/NTAG213_215_216.pdf) is mentioned that command. 
In fact, the command always return -20 (rf timeout error). 
Here you can see the libnfc log:

debug   libnfc.driver.pn53x_usb TX: 00 00 ff 04 fc d4 42 **60 00** 8a 00 
debug   libnfc.driver.pn53x_usb RX: 00 00 ff 00 ff 00 
debug   libnfc.chip.pn53x       PN53x ACKed
debug   libnfc.driver.pn53x_usb RX: 00 00 ff 03 fd d5 43 **01** e7 00 
debug   libnfc.chip.pn53x       Chip error: "Timeout" (01), returned error: "RF Transmission Error" (-20))

What am I missing here?
Thanks

SloCompTech, 2019-06-07:
Hi,
yeah, maybe I gave function a little improper name, basically it is function GET_VERSION (if you look at the docs), and only purpose is to differentiate between ntag21x and ultralight tags

shad0w82, 2019-06-10:
Ok i see, but then what's the difference with the function "ntag21x_get_info" (that according to the source code, it's rightly used to retrieve the information on the tag, as per the datasheet)? 
More over, the "ntag21x_is_auth_supported", apart from having a little improper name, is always going to fail, because the bytes 60 00, is not matching any valid command...

Gf

SharonovMaxim, 2019-07-31:
> Ok i see, but then what's the difference with the function "ntag21x_get_info" (that according to the source code, it's rightly used to retrieve the information on the tag, as per the datasheet)? [...]

 Hey. I have the same issue Chip error: "Timeout" (01), returned error: "RF Transmission Error" (-20)). But I'm just trying to use nfc-list in Kali linux. I don't know if it's a hardware issue. I want to know how it goes in your case. My device is pn532 -> UART USB

shad0w82, 2019-08-02:
Actually I ended up rewriting my own version of libfreefare, with only the features I needed, in QT.

If you are trying to use the nfc-list, the problem is a hardware one. Me too I had that same problem on one pc, but not on one another..I’m using a PN533 with native usb Interface.

 

Gf

MaxMaxov93, 2019-08-02:
Ok, thank you for the answer. I have thought me device (pn532 chip) came broken. But it seems to be something software. Or is it a hardware issue that has to do with pc, not the pn532 itself?

shad0w82, 2019-08-02:
No I think the PN532 it’s ok. My problem was related to the PC I was using.

Gf

darconeous, 2019-10-29:
Closing to clean up issues. Re-open if closed in error.

closed 2019-10-29 by darconeous

realhik, 2020-07-09:
I had the same issue. I think the command ist just "0x60", not "0x60 0x00" for GET_VERSION.