nfc-tools/libfreefare #114: ARM issue desfire_crc32 wrong direction

alenloncaric, 2019-08-20:
Dear,

 found an issue how arm interprets direct assignment of crc in the buffer
       *((uint32_t *)(crc)) = htole32(desfire_crc);
On non-arm this does as expected at the position of crc pointer CRC is added to the right side of the buffer crc, crc+1, crc+2, crc+3

On the ARM that we are using things are opposite crc is added to the left of the buffer crc, crc-1,crc-2,crc-3 :)

We corrected the code with memcpy: 
   memcpy(crc, &htole32(desfire_crc), sizeof(uint32_t));

Hope this helps someone

smortex, 2019-08-20:
Uh…  Another option would be to use binary operators to make all this obvious:

```c
crc[0] = desfire_crc & 0x000000ff;
crc[1] = (desfire_crc & 0x0000ff00) >> 8;
crc[2] = (desfire_crc & 0x00ff0000) >> 16;
crc[3] = (desfire_crc & 0xff000000) >> 24;
```

(unless it is the other way arroud…).  Maybe a bit more readable and I would be surprised if the compiler does not optimize both snippets the same way.

Would you mind filling-in a Pull Request?

darconeous, 2019-10-29:
> On the ARM that we are using things are opposite crc is added to the left of the buffer crc, crc-1,crc-2,crc-3 :)

I'm guessing the pointer `crc` isn't aligned on a 32-bit boundary. Classic undefined-behavior bug.

What file is this in?

denisdemaisbr, 2023-05-01:
smells an endianness issue