nfc-tools/libfreefare #116: Support for NTAG 424, DESFire Lite, and some limited DESFire EV2

darconeous, 2019-10-04:
`libfreefare` currently supports the legacy *MIFARE DESFire D40* and *MIFARE DESFire EV1*. It supports EV2 only to the extent that EV2 is able to function identically to a EV1, but no EV2-specific functionality is currently supported.

This is complicated by NXP's requirement that the full datasheet for the DESFire EV2 only be released under NDA. The [short datasheet](https://www.nxp.com/docs/en/data-sheet/MF3DX2_MF3DHX2_SDS.pdf) is not adequate to implement support for the new features such as the new EV2 security mode, ability to update arbitrary records, advanced key management, proximity detection, etc.

BUT...

NXP has recently released two new chips that use a subset of the EV2 command set:

 * [NTAG 424 DNA][] ([NT4H2421Tx][])
 * [MIFARE DESFire Light][] ([MF2DLHX0][])

Critically, *both of these chips have the full datasheets available*! 🎉

By carefully reading between the lines from both datasheets, we should have enough information to implement the following features:

* [ ] [EV2-Style Security Sessions][] (Both AES and [LRP][])
* [ ] [EV2 TransactionMAC][]
* [ ] Full ATS customization
* [ ] Support for the [UpdateRecord][] command, which updates data of an existing record in the Cyclic Record File.
* [ ] Authentication attempt limits/throttling

It's unclear if there is enough information in the datasheets to piece together how the multiple-keys-per-access-right feature works, but that might be possible, too.

In addition, the [NTAG 424 DNA][] is itself a pretty cool chip:

* Up to 5 application-specified AES encryption keys.
* Secure data file separate from NDEF storage.
* [Secure Dynamic Messaging][], which allows you to create what amounts to an NDEF authentication token.
* 50 year data retention length is considerably larger than that of any DESFire chip, making it suitable for reliably storing long-term secrets.

So, by adding support for the [NTAG 424 DNA][] and [MIFARE DESFire Light][], we end up getting *some* support for a few DESFire EV2 features, even though we don't have access to the full datasheet for that chip. 

--------

So I'm considering adding support for these chips. Neither are very common at the moment. I'd consider adding the NTAG 413 DNA as well, but it's datasheet is only available under NDA.

Thoughts? If I do this, do you have any guidance/opinions on how the API should work?

[NTAG 424 DNA]: https://www.nxp.com/products/rfid-nfc/nfc-hf/ntag/ntag-for-tags-labels/ntag-424-dna-424-dna-tagtamper-advanced-security-and-privacy-for-trusted-iot-applications:NTAG424DNA
[NT4H2421Tx]: https://www.nxp.com/docs/en/data-sheet/NT4H2421Tx.pdf
[MIFARE DESFire Light]: https://www.nxp.com/products/rfid-nfc/mifare-hf/mifare-desfire/mifare-desfire-light:MIFARE_DESFIRE_LIGHT
[MF2DLHX0]: https://www.nxp.com/docs/en/data-sheet/MF2DLHX0.pdf
[Secure Dynamic Messaging]: https://www.nxp.com/docs/en/data-sheet/NT4H2421Tx.pdf#page=35
[LRP]: https://www.nxp.com/docs/en/application-note/AN12304.pdf
[EV2-Style Security Sessions]: https://www.nxp.com/docs/en/data-sheet/MF2DLHX0.pdf#page=22
[EV2 TransactionMAC]: https://www.nxp.com/docs/en/data-sheet/MF2DLHX0.pdf#page=39
[UpdateRecord]: https://www.nxp.com/docs/en/data-sheet/MF2DLHX0.pdf#page=102

darconeous, 2019-10-04:
And while doing that, I might be able to swing fixes for both #90 and #91.

SirNiggo, 2019-11-06:
Is NXP really trying to have security through obscurity? Didn't they learn from their crypto-1 faults? Btw, thanks for doing such a great work for the community. (Y)

darconeous, 2019-11-06:
NXP’s designs don’t rely on proprietary crypto anymore.

If anything I think they are keeping their specs under NDA because they don’t want to help people make knockoff chips. But that’s just a guess. 

It is frustrating, but I think they’ve learned their lessons from the MIFARE Classic days.

doegox, 2019-11-06:
Desfire EV2 is a EAL5+ certified product and when you do such CC certification, limiting accessibility of information that could potentially help attackers is valued.
That's the rules of the game.

darconeous, 2019-11-06:
Is that really one of the CC EAL5+ requirements? If so, that is pretty lame.

But I don't see any obvious correlation between CC EAL certifications and NDA requirements, aside from the "+":

Product | CC Cert | NDA Required
--------|----------|---------------
DESFire EV2 | EAL5+ | YES 
DESFire EV1 | EAL4+ | YES 
DESFire Lite | EAL4 | NO
NTAG 413 DNA | None | YES
NTAG 424 DNA | EAL4 | NO

doegox, 2019-11-06:
It's not requirements, it gives you extra points and you need to reach some amount of points to get your certification. So it's not impossible to be more open and get certified, but it's harder. And going from EAL4+ to EAL5+ is already pretty hard to achieve...

darconeous, 2019-11-06:
How unfortunate. I was under the impression that EAL certifications were more about design practices and confirmability/testing than gimmicks like limiting access to the specification to a software interface.

In any case, it is what it is, and I doubt it will change—but I was pleasantly surprised that both the NTAG 242 DNA and DESFire Lite specifications don't need NDAs. I managed to pick up a few NTAG 242 DNA tags a few weeks ago.

icedevml, 2019-12-29:
Hello guys. I have been challenging the EV2 just today in order to implement a library in Python 3. Could say some examples in the application notes are somehow "internally inconsistent".

I think it may be useful to you guys, I have working test cases for AuthenticateEV2First and CommMode.MAC/FULL:

https://github.com/icedevml/ntag424-ev2-crypto/blob/77efa9f8bc217b95e45a93e456570eaa1f337f86/ev2.py#L211

Also there is my implementation in the same file, may save you some time if you look at it.

Unfortunately right now I don't have time to port it to libfreefare and I don't even have the devices physically on site to test anything.

Silur, 2020-06-15:
https://github.com/islog/liblogicalaccess support the DesFire EV2 I'm pretty sure we can learn stuff there

p-gentili, 2021-03-16:
Hi everyone, I'd like to contribute to this project adding support for the ntag424. I've been working with it for my job so I think my help could be useful. What's the current status of the development?

astrand, 2021-05-14:
Is it possible to use NTAG 424 DNA tags with libfreefare (using only basic functionality), with the NTAG21x support? In other words, are NTAG424 tags backwards compatible with NTAG21x?

icedevml, 2021-05-14:
@astrand 424 and 21x have very very little in common:

* NDEF content is accessed in a different way (NFC Forum Type 4 Tag vs NFC Forum Type 2 Tag)
* Configuration command set is different
* Mirroring is configured in a different way

I think these two need separate support.

rena2019, 2023-01-04:
seems that [https://github.com/icedevml/ntag424-ev2-crypto](https://github.com/icedevml/ntag424-ev2-crypto/blob/be951150c54d2f6846bca57ce76b33ad3412acdc/ev2.py#L233) is not available anymore?