nfc-tools/libfreefare #121: Cut a 1.0.0 release

darconeous, 2019-11-06:
We should consider cutting a 1.0.0 release soon.

Before we cut a 1.0.0 release, we need to do the following:

* [ ] Resolve all existing security issues (like #33 and #90)
* [ ] Update the version information in `configure.ac`
* [ ] Increment the library compatibility version.
* [ ] Review all of the APIs and make sure we are satisfied with how they work.
* [ ] Designate who will sign the release, or what key to use to sign the release.

smortex, 2019-11-06:
:+1:

I am not 100% sure but AFAICR, building libfreefare master branch against the latest libnfc release does not work.  So maybe we should also release a new libnfc version in the process.

fuzxxl, 2020-01-05:
Please consider removing pragma pack from all structures exposed in headers.  You can do this for free since you already break the API in this new major release.  This makes writing bindings to other languages a lot easier.

fuzxxl, 2020-05-21:
The libnfc has meanwhile pushed an 1.7.2 version with a bunch of new APIs.  Might be a good idea to push a new libfreefare version in tandem.

mtdcr, 2020-09-14:
>  Review all of the APIs and make sure we are satisfied with how they work.

Functions like `mifare_desfire_read_*` don't know the size of the `data` buffer passed to it. If `length` is zero, the whole file will be read into this buffer. I don't know what's the maximum size of a file from the top of my head, and whether it's mandatory to pass a buffer with maximum file size when `length` is zero, but it's something I suspect many people can get wrong.

Therefore I suggest adding a size parameter to all functions writing into a buffer provided by a user of the library.

clausecker, 2021-07-23:
Any progress on this one?

graynk, 2021-09-10:
I'd like to remind everyone that there are Buildroot pipelines that are dependant on libnfc/libfreefare, but due to Buildroot's nature it downloads the latest _release_.  Thankfully, libnfc's 1.8.0 is in somewhat OK state to use now, but libfreefare is still 6 years old at this point.

I had to .tar.gz the repo and upload it to my own hosting, adding `LIBREEFARE_AUTORECONF = YES` to Buildroot's config. It's not _that_ much of a problem, but it's certainly _a_ problem.