nfc-tools/libfreefare #122: Create a 0.4.x branch

darconeous, 2019-11-06:
We should consider creating a 0.4.x branch with backported security fixes.

The 0.4.x branch would allow existing applications that require the original API to continue to work, but still be able to receive security updates and changes that don't break the established 0.4.x API/ABI.

I'd recommend branching at faac4ae5d881a12d7fb81fc831629ae1d1f10929 (See #46), since that is the last commit before breaking API changes occur.

Maintenance of the 0.4.x branch would be on a best-effort basis.

smortex, 2019-11-06:
No objection.  Please tell me if your current privileges do not allow you to do this.

clausecker, 2022-10-20:
Any progress on this one?

clausecker, 2022-10-26:
Hi @smortex.  I'll look into what I can do.