nfc-tools/libfreefare #123: Wiki, Website and Forum not available

SirNiggo, 2019-11-06:
Hello,

I "accidentally" found this awesome project today and wanted to find out more about the whole project. Sadly, the wiki, website and forums do not seem to be available right now.
Is this intended? I am trying to access the sites from Germany. If I can help you out in any way please tell me how I can.
If you need a small free virtual machine for hosting those sites I am sure that we can help out. We do have a machine in a datacenter in Frankfurt (Germany) with enough power to host some more VMs and a gigabit connection. If free hosting is needed, I would greatly appreciate to help you out with that (limited to project specific functionality).

Kind Regards
Nico

darconeous, 2019-11-06:
This is a known issue: <https://github.com/nfc-tools/libnfc/issues/558>

Unfortunately it has been down for quite some time. Like *years*.

The [whole project](https://github.com/nfc-tools) needs a lot of love. I'm only participating in the `libfreefare` side of things.

But, ultimately, we should consider removing the links if this isn't something that is going to be resolved soon.

SirNiggo, 2019-11-07:
Thank you. Closing this as its a duplicate.

closed 2019-11-07 by SirNiggo