nfc-tools/libfreefare #129: Specs for setting encrypted transfer modes with DESFire tags

maxieds, 2020-06-17:
I'm not sure where else to ask this question. You all seem to have a good working implementation of the crypto routines for the Mifare DESFire tags. This issue is obviously complicated by the fact that inner details of these tags tend to run by NDA-only standards with the manufacturers. I have a question about clarifying when to (can I) invoke certain encrypted transfer modes with these tags. 

Authentication with 2KTDEA/3KTDEA/AES is well enough documented to me. Where does the communication mode for transferring data (in general, for all commands and data, not just specific files) get specified in communicating with these tags? I cannot seem to find a solid reference (or data sheet or application note) for where this assumption is made in a clear cut way across the EVx standards. 

Historically, there should be three (plaintext, 2KTDEA, 3KTDEA) and more recently there are AES communication modes in key sizes of 128/192/256. One possibility is to just attach the current communication mode to whatever the most recent Authenticate instruction uses, but that is probably not up to standard, and is somewhat arbitrary. So my question is, besides just deciding one standard over the other by default, how can you tell whether the transfer sessions from ``PICC<->PCD`` should be with 3DES, or AES, and if so which key sizes should get used and when?

tlo2357, 2021-02-05:
If not for a bug, mifare_desfire_get_key_settings() would return two bits that indicate the encryption type of the application's keys, for an EV1 at least.  See issue #135.

smortex, 2021-03-04:
@maxieds the answer provided @tlo2357 seems to be the right way to go.  I think we can continue the discussion there, so I will close this issue.

Thanks!

closed 2021-03-04 by smortex